elcrni/Voron-Mods #2: Clearance for heatset tool

eddietheengineer, 2022-09-26:
Hello @elcrni! Thanks for making this mod, it looks really well done! 

I've started printing the parts, but when I tried to insert the heatsets into the front piece, there wasn't enough clearance for the heatset insertion tool (I'm using LDO's), so it ended up melting the plastic:

[image]

[image]

Also, consider opening up the back screw hole pass throughs for the NEMA14 motor mount! This would allow mounting of M3 standoffs to mount a CAN toolhead board such as the EBB 36 V1.2

[image]

elcrni, 2022-09-26:
@eddietheengineer thanks for the feedback. I am aware of the issue but for now that's the only way. I am using thinner soldering tip and have no issues and not touching the plastic. Will see if this can be avoided completely. many thanks

mstawick, 2022-09-26:
@eddietheengineer FYI, maybe it's helpful to you, I've used these tips:
https://www.amazon.pl/dp/B09WPJ56JP/ref=pe_25095781_587317651_TE_item

and was able to embed the inserts cleanly.

elcrni, 2022-09-26:
> @eddietheengineer FYI, maybe it's helpful to you, I've used these tips: https://www.amazon.pl/dp/B09WPJ56JP/ref=pe_25095781_587317651_TE_item
> 
> and was able to embed the inserts cleanly.

Awesome, thanks for the link!

closed 2022-10-03 by elcrni