elcrni/Voron-Mods #4: ERCF version

xbk666, 2022-10-06:
I really like your design and currently use it without any problem. However I am planning to build my ERCF and would really appreciate there could be a design compatible with ERCF's toolhead sensor. Thank you in advance.

elcrni, 2022-10-06:
thanks for the comment, i will certainly have that in mind for the next release, very soon.

xbk666, 2022-10-10:
> thanks for the comment, i will certainly have that in mind for the next release, very soon.

Thanks in advance, looking forward to it

TravisWilder, 2022-11-22:
Looking also forward for a sensor version.
the PCB Version in the Step seems also to miss some kind of spacer (between PCB and cable cover) and I do not see room for the screwhead. may be you can let us know how this shall work :)

A version for the new PCB (with the small PCB on the FAN) will anyway impossible as the Orbiter will need the space

elcrni, 2023-01-12:
This is work in progress now

hadesz, 2023-01-14:
I'm half way through with the toolhead sensor model, I can open a PR or something... but I need to change both the front and back part of the mount to make room

suizhenmin, 2023-01-23:
It is more convenient to use steel balls and micro switches for ERCF tool heads than to use Hall sensors, and it is relatively less difficult to design.Looking forward to it, I'm also assembling ERCF.

ndng28, 2023-01-25:
> I'm half way through with the toolhead sensor model, I can open a PR or something... but I need to change both the front and back part of the mount to make room

Let me know if you need help testing it.

hadesz, 2023-01-28:
I need to make some adjustments:
 - the filament whole is slightly too big
 - the hall effect sensor is about 0.1mm further than needed

But I had an accident today while investigating why the 5V rail does not provide 5V, so waiting for a new MB. I can share untested STLs, if someone is interested, otherwise I can test next weekend and and publish after.

> It is more convenient to use steel balls and micro switches for ERCF tool heads than to use Hall sensors, and it is relatively less difficult to design.Looking forward to it, I'm also assembling ERCF.

I will look into the BTT Laser sensor later as well, but as I know ERCF can work sensorless as well

suizhenmin, 2023-01-29:
> I need to make some adjustments:
> 
> * the filament whole is slightly too big
> * the hall effect sensor is about 0.25mm further than needed
> 
> But I had an accident today while investigating why the 5V rail does not provide 5V, so waiting for a new MB. I can [share untested STLs](https://github.com/hadesz/Voron-Mods/tree/main/Orbiter_2.0_SB_CW2_Enclosed/STL), if someone is interested, otherwise I can test next weekend and and publish after.
> 
> > It is more convenient to use steel balls and micro switches for ERCF tool heads than to use Hall sensors, and it is relatively less difficult to design.Looking forward to it, I'm also assembling ERCF.
> 
> I will look into the BTT Laser sensor later as well, but as I know ERCF can work sensorless as well

It's great, I'll print a test piece to have a look later, it would be even better if it can be modified to add support for Orbiter Filament Sensor Summary.

suizhenmin, 2023-01-30:
> I need to make some adjustments:
>
> * the filament whole is slightly too big
> * the hall effect sensor is about 0.25mm further than needed
>
> But I had an accident today while investigating why the 5V [...]

I tested it and your modification is successful, but the position of the hall sensor is very difficult to control.

ndng28, 2023-01-30:
> I need to make some adjustments:
>
>     * the filament whole is slightly too big
>
>     * the hall effect sensor is about 0.25mm further than needed
>
>
> But I had an accident today while investigating why [...]

I have been waiting for the hallsensor to be shipped and it looks like it might not happen anytime soon.  Appreciate your effort here...would you consider making a micro switch+ball version? Based on the discord conversations, it seems like the microswitch version is more accurate than hall effect.

hadesz, 2023-01-30:
> 
> I tested it and your modification is successful, but the position of the hall sensor is very difficult to control.

It is the same design as Voron CW1, I use the very same for my previous version of orbiter mount, where it was easier to assemble, maybe I can make the hole a bit bigger, but the space is very limited for alternative solutions

> 
> I have been waiting for the hallsensor to be shipped and it looks like it might not happen anytime soon. Appreciate your effort here...would you consider making a micro switch+ball version? Based on the discord conversations, it seems like the microswitch version is more accurate than hall effect.
> 
> Something to this effect: https://github.com/EtteGit/EnragedRabbitProject/tree/main/usermods/CW2_with_ECAS_and_optional_sensor
> 
> Thanks for considering.

I can give it a try, but I have no HW to test it

suizhenmin, 2023-01-31:
> It is the same design as Voron CW1, I use the very same for my previous version of orbiter mount, where it was easier to assemble, maybe I can make the hole a bit bigger, but the space is very [...]

The Hall sensor is very difficult to use. The magnet is too strong, which will cause the Hall sensor to always sense, no matter how you move the position. The magnetism is weak, and it cannot be sensed. I have tried 3 different magnets and different combinations such as using 2 pieces, but I still can't adjust it until it can sense correctly. I'm in China and it's very easy to get these accessories. In the end, I gave up the solution of using the Hall sensor, hoping to design a solution using steel balls and micro-movement, which is simpler and can be realized with only 2 wires.

hadesz, 2023-01-31:
> The Hall sensor is very difficult to use. The magnet is too strong, which will cause the Hall sensor to always sense, no matter how you move the position. The magnetism is weak, and it cannot be [...]

As per ercf manual,  the sensor and magnet must be all the way in,  but the orientation is important to work properly.  No need for 2 magnets, or stronger ones... what I can imagine messing with the setup are the screws holding the front and rear together.  I have my machine running again (new board and drivers), I'll test and fine tune it on the weekend

suizhenmin, 2023-01-31:
> As per ercf manual, the sensor and magnet must be all the way in, but the orientation is important to work properly. No need for 2 magnets, or stronger ones... what I can imagine messing with the [...]

The orientation of the hall sensor and the magnetic pole are correct, this test. In principle, it is the magnetic conduction of the magnet to the iron gasket, and then when the filament is loaded, the iron gasket is pushed away, and the Hall sensor senses the iron gasket that conducts the magnetism. It is very difficult to determine the position of the Hall sensor at that time.

hadesz, 2023-01-31:
> The orientation of the hall sensor and the magnetic pole are correct, this test. In principle, it is the magnetic conduction of the magnet to the iron gasket, and then when the filament is loaded, [...]

I used the very same bodies to cut the holes in this CW2 mount that I used for the Orbitu CW1 mount where it works perfectly, the only difference are the metal screws holding back and front together closer to the sensor, what can change the magnetic field, I will do other versions in the weekend, but if you can just test it without the screws,  it would help

hadesz, 2023-02-03:
> But I had an accident today while investigating why the 5V rail does not provide 5V, so waiting for a new MB. I can [share untested [...]

So my machine is back up again, and I can confirm, the close proximity of the 2 screws throw off the sensor, i'll check if I can relocate them

hadesz, 2023-02-05:
I did some testing, and changed the alignment og the sensor, b4 version is what I will mount, but the filament hole was a tad small (used an 2mm drill), the only change in b5 is the fix for this
And realigned the folder structure to follow the other Voron mods structure: https://github.com/hadesz/Voron-Mods/tree/main/Orbiter_2.0_SB_CW2_Enclosed/STL/alternatives/hall_sensor

ndng28, 2023-03-17:
I can confirm that the hall sensor mod here (version 5) works really well!

Question - how long is your PTFE tube coming out of the hotend?

fryc88, 2023-07-21:
> I did some testing, and changed the alignment of the sensor, b4 version is what I will mount, but the filament hole was a tad small (used an 2mm drill), the only change in b5 is the fix for this And realigned the folder structure to follow the other Voron mods structure: https://github.com/hadesz/Voron-Mods/tree/main/Orbiter_2.0_SB_CW2_Enclosed/STL/alternatives/hall_sensor

Any clues how to build it up and some BOM for it?
I'm about to reprint this enclosed project but i'm not sure what else i would need for filament sensor.

hadesz, 2023-07-21:
> > I did some testing, and changed the alignment of the sensor, b4 version is what I will mount, but the filament hole was a tad small (used an 2mm drill), the only change in b5 is the fix for this And realigned the folder structure to follow the other Voron mods structure: https://github.com/hadesz/Voron-Mods/tree/main/Orbiter_2.0_SB_CW2_Enclosed/STL/alternatives/hall_sensor
> 
> Any clues how to build it up and some BOM for it? I'm about to reprint this enclosed project but i'm not sure what else i would need for filament sensor. And do you know if it won't interfere with latest changes what original author made? I know mainly it was hole sizes that i don't know if you have changed, but it would be nice to know before printing it all over again :)

On top of the standard bom, the sensor needs a toolhead sensor kit like [this](https://www.aliexpress.com/item/1005004795802987.html)

fryc88, 2023-07-21:
By any chance have you got any photos how does that look like with hall sensor and magnet mounted? And do i need any kind of shaft for it or it's just magnet + hall sensor? I've seen few different approaches to hall filament runout sensors but i never used one, so i'm not 100% sure how it should be put together really.
I do have some parts, including shafts, magnets and hall sensors, so i should be able to build it, but it would be easier with just some photo :)
Really appreciate your quick reply!