elcrni/Voron-Mods #5: Version with EBB 36 CAN V1.1 Holder

pforrmi, 2022-10-08:
Hi,

first thanks very much for this amazing design! I really like it.
I'm using the Stealtburner with the BIGTREETECH EBB 36 CAN V1.1. Is it possible you can adjust it to mount it at the toolhead?

Bests
Michael

ReXT3D, 2022-10-29:
FWIW, I am targeting the same tool head configuration using Stealthburner, Orbiter 2 & EBB36 (or alternatively EBB42).  I have defined what I believe is a reasonably optimal design configuration, but it would require shifting the complete Orbiter 2 assembly up by about 6 mm in order to clear interference highlighted below:

[image]

From design requirements perspective, I prefer to retain the chain and I feel very strongly that active cooling is required for the stepper motor and for the EBB36 in order to improve long term reliability in a heated chamber.  The above sample layout moves the chain end point by the length of two links and includes a location for 3010 cooling fan.

ReXT3D, 2022-10-29:
Just had another look.  Placing the EBB36 orthogonally would indeed work much better and it appears to need only reasonably small changes to your original design:

1. Revised **Chain Anchor** for new chain end location and to support one of the two EBB36 mounting points
2. Revised **Rear Body Orbiter 2** to support the second EBB36 mounting point with minimal fan flow obstruction
3. Revised **Cable Cover for PCB** to support installation of 3010 fan

[image]

Unfortunately I am still in the process of building my 2.4 and it would be a while before I can have a go at making these changes.

ReXT3D, 2022-11-06:
Another feasibility concept with EBB42.  It is awfully tight in there, and overall the "stealth" surface geometry makes it a bit challenging to work with TBH...

[image]

pforrmi, 2022-11-06:
I really like your designs! Thanks for sharing.

I'm using the EBB36 for now without a cable chain. Also thinking that a fan is not necessary for cooling cause I printed some hours with enclosed 2.4 and no bad things happened with inside temperature at about 50 degrees.

Do you uploaded your designs yet?

ReXT3D, 2022-11-06:
> Do you uploaded your designs yet?

Thank you for your comments.  Note that this only a proof of concept & fitment check and not a design.  I just played with F360 using author's supplied model to identify potential installation concepts that would meet my requirements.

As I mentioned before, I feel very strongly about the need for forced cooling of the stepper and the toolboard.  Orbiter 2 stepper runs very hot even on my unenclosed CR-10S Pro in a cool basement.  Having the stepper additionally enclosed with plastic brackets & covers and in a heated chamber will bring its temperature very close to ABS deflection temperature and likely above ABS+ glass transition.  In addition, a lot of the stepper heat is also transmitted into the Bondtech gears, which is not ideal for the filament.  Finally, electronics failure rate (unreliability) increases exponentially with operating temperature and therefore you always want to keep the electronics as far away from their design limits as possible.  And I doubt that BTT (or Mellow for that matter) have done any thorough thermal derating analysis of their boards...

mstawick, 2022-11-07:
FYI, on the orbiter project page (https://orbiterprojects.com/orbiter-v2-0/), you can expand the firmware section, and there's the interesting part under `3.7 Motor current vs extruder performance and temperature`. I've run some tests with NTC glued to the orbiter motor, and the provided numbers seem to be accurate (not that I would assume them not to be, but better test and be safe than sorry...).

This means that in an enclosure, it's either active cooling, reduced current, or melted Stealthburner ;)

elcrni, 2022-11-15:
Thank you for your efforts and an interesting solution. 
I am now working on adding the CAN board. Your solution for the fan and the cable cover looks nice, but are you cooling the motor or the board with that variant? 
Perhaps you could share your files, cable cover that is.
many thanks

> Just had another look. Placing the EBB36 orthogonally would indeed work much better and it appears to need only reasonably small changes to your original design:
>
> 1. Revised **Chain Anchor** for new [...]

ReXT3D, 2022-11-15:
> Your solution for the fan and the cable cover looks nice, but are you cooling the motor or the board with that variant?
> Perhaps you could share your files, cable cover that is.
> many thanks

Hello, I am glad to see that you are able to work on the CAN board implementation.  As I noted earlier to **pforrmi** this was really just a form & fit proof of concept rather than a design.  I was trying to figure out what placement would be feasible and if things could fit...

With respect to the fan, it can be placed to cool both, the stepper body and the back side of the EBB board (correctly installed TMC drivers dissipate the heat through the back side into the PCB).  In my attempt to place the fan with minimal changes to the cover, I picked this potential position:

[image]

But if one were to make larger scope changes to the cover, the fan could be moved another 1-2 mm towards the back of the printer to increase the EBB back-side cooling.

ReXT3D, 2022-11-18:
I am attaching the cover concept STEP file in case you find it useful for positioning the fan.  It also includes a 3010 fan model that I put together a while ago from bits and pieces I found online.  As I said before, this is just a concept and the cover needs detailed design work to allow fan attachment, etc.

[Fan Cover Concept.zip](https://github.com/elcrni/Voron-Mods/files/10043202/Fan.Cover.Concept.zip)

pireybr, 2022-12-12:
> I am attaching the cover concept STEP file in case you find it useful for positioning the fan. It also includes a 3010 fan model that I put together a while ago from bits and pieces I found online. [...]

Please release a final version of it :)

MLeal91, 2022-12-21:
waiting for this too, orbiter is awsome and with canbus is the best but at momment dont have any mount to can with orbiter. will wait for this

elcrni, 2022-12-28:
Hi all, 

1st revision of the EBB36 CAN mount is available for testing now. Keep in mind, there is no cable chain mount, only the PG7/Umbilical mount, for now.

Files are here:
https://layerguru.com/wp-content/uploads/2022/12/EBB36_SB_Orbiter_R1.zip

I would appreciate the feedback. Many thanks.

[image]
[image]

MLeal91, 2022-12-28:
> Hi all,
>
> 1st revision of the EBB36 CAN mount is available for testing now. Keep in mind, there is no cable chain mount, only the PG7/Umbilical mount, for now.
>
> Files are here: [...]

it is something now :) simple design and functional .... you will work on solution to use chain?

elcrni, 2022-12-29:
> it is something now :) simple design and functional .... you will work on solution to use chain?

Yes, cable chain version will follow.

MLeal91, 2023-01-02:
just wait for usb cable to connect this :)
Cable Cover for PCB 3010 fan is for future idea? i like idea of cooling stepper motor and can

ndng28, 2023-01-10:
Hey thanks for working on this. Some feeedback...the CANbus mount oscillates during input-shaper tests and I am getting some wild data. I think it needs a third stand-off screw or something to provide additional rigidity.

elcrni, 2023-01-11:
> Hey thanks for working on this. Some feeedback...the CANbus mount oscillates during input-shaper tests and I am getting some wild data. I think it needs a third stand-off screw or something to provide additional rigidity. Also the CANBus needs to be mounted perpendicular to the axis.
> 
> https://github.com/KayosMaker/CANboard_Mounts

Yes, i am aware of the ADXL issue as the board is rotated 90 degrees but that's pretty much the only way to mount it. The reason you are getting wild IS data is the rotation of the board and not the mount itself.

closed 2023-01-11 by elcrni