elcrni/Voron-Mods #6: LDO Toolhead PCB support

PuszInPusz, 2022-11-26:
Dear creator, do you think there is a way to incorporate the LDO toolhead pcb in your design? Sofar your Orbiter design is the best I saw but I don’t think it’s compatible with my Ldo toolhead.

arfr, 2022-12-07:
@PuszInPusz please take a look at https://github.com/MotorDynamicsLab/LDOVoron2/tree/main/STLs/LDOSBCW2MountCover they adopted it, maybe you can do it too :)

bjensen, 2023-01-03:
@PuszInPusz How did you end up mounting the LDO toolhead?

PuszInPusz, 2023-01-03:
@bjensen Hey, I didn’t. Before the holidays I designed something but I didn’t like it because I couldn’t close the lid. Between the holidays I did nothing but I’ll maybe begin to redesign my first attempt next week.

bjensen, 2023-01-03:
@PuszInPusz care to share your design? As @arfr pointed out it might be easier to increase the size of latch if you have made a good mount

elcrni, 2023-01-03:
guys, LDO toolhead PCB should be taken from hartk, so identical to hartk SB PCB... in this case i have already made the mount for that PCB. just look under PCB topic

arfr, 2023-01-03:
where?

elcrni, 2023-01-11:
There are no plans to incorporate LDO toolhead PCB at the moment. HARTK1213 PCB can be used instead (Single piece PCB only)

closed 2023-01-11 by elcrni