elcrni/Voron-Mods #7: PCB Support

Fragmon, 2022-12-23:
Hey, i see that you include the pcb in your step but you dont make a mount for this board. Do you already do this or have i remix your files for a pcb mount?

elcrni, 2022-12-28:
Hi there, there is still no support for PCB due to the design of the Orbiter extruder itself, its very very tricky to place the toolhead PCB under the cover, i am still testing different variants but can not give any ETA. thanks

elcrni, 2022-12-28:
PCB support is now in testing stage. The PCB has to go further back compared to CW2, but this should not be an issue.

Initial release files will be available soon for testing.

[image]
[image]
[image]

elcrni, 2022-12-28:
Beta PCB mount files can be downloaded from here:
https://layerguru.com/wp-content/uploads/2022/12/Orbiter_2_SB_PCB_mount.zip

Feedback would be appreciated.

PuszInPusz, 2023-01-03:
@elcrni is there maybe a way you do a LDO PCB support? I myself tried to remix but I have the same issue. Can’t manage to close the lid such that the „contouring“ stays seamless.

bjensen, 2023-01-06:
@elcrni Can you please release the cad files? Its hard to see how the pcb mount is attached to the rest..

Gaschtae, 2023-01-10:
Can somebody please reupload the Files? The website seems to be down right now.

elcrni, 2023-01-10:
gloval outage for SiteGround servers, should be back online shortly

Fragmon, 2023-01-10:
works for me. Only problem is, that the cable from the mainboard crash with the z chain when he goes from the y,x endstopp to the z endstopp. I solve this problem with an a y move after x,y to get space between the chain and the cable.

elcrni, 2023-01-11:
> works for me. Only problem is, that the cable from the mainboard crash with the z chain when he goes from the y,x endstopp to the z endstopp. I solve this problem with an a y move after x,y to get space between the chain and the cable.

Unfortunately this is as far as the PCB can go forward. Some evade code may be necessary to prevent Microfit connector and wires touching Z chain as @Fragmon did.

closed 2023-01-11 by elcrni

damianhw, 2023-02-25:
Does anyone know of a version compatible with 2 piece PCBs like the hartk one or the SB2240/2209 Canbus one? Not sure how it could work with the tension lever in the way though.

elcrni, 2023-02-25:
> Does anyone know of a version compatible with 2 piece PCBs like the hartk one or the SB2240/2209 Canbus one? Not sure how it could work with the tension lever in the way though.

No, this is not possible, not just because of the lever but the fact that the front side of the orbiter is touching the part cooling fan, simply no space to place the fan PCB.