elcrni/Voron-Mods #8: Misc issues with Orbiter 2.0 mount

bjensen, 2023-01-06:
First off: Thank you for making the mount for the orbiter 2.0. 

I have assembled it and I have some findings: 

1. The piece that is touching the Orbiter top, right where the entry of the filament is super snug. The ASA Im using is almost bent a little. I have tried to attached some images that displays what Im talking about:

[image]
[image]

2. More details of which screws to use where. For example when looking at https://github.com/elcrni/Voron-Mods/blob/main/Orbiter_2.0_SB_CW2_Enclosed/Images/Orbiter_2.0_SB_CW2_Enclosed__006.jpg its not possible to see which mm screw should be used. 

3. More details on how to mount the PCB mount. Ive printed out the beta pieces but cannot see how it is attached, even though Ive been looking at the images in the PCB support thread. Perhaps if you were to include an image where the latch and the pcb is removed it would be obvious to see.

Thanks again

elcrni, 2023-01-06:
hey, thanks for the feedback.

1. Yes, its a little snug but the goof thing is that the front body part can be shaved off a little for easier and roomier fit. this will be done in next iteration soon,
2. unfortunately i don't always have the time to annotate everything, hopefully will sort that out as well soon. As first aid, just do trial and error with the screw length :-)
3. image of the PCB mount attached, again lack of time to do everything but soon.
[image]

hope this helps, thanks.

bjensen, 2023-01-06:
Thanks for your prompt reply. How about releasing the cad files? 

That is quickly done and i will be able to make a pull request with my suggested changes and see how things to together and so will others.

elcrni, 2023-01-06:
soon, when the next revision is ready, by monday evening i hope

closed 2023-01-11 by elcrni