elcrni/Voron-Mods #9: Any thoughts on supporting the Filament Sensor addon?

dkowis, 2023-01-07:
https://orbiterprojects.com/orbiter-filament-sensor/

I know it'd blow the cover entirely out of the water, but I'm kinda okay with that. Thought I'd ask about it, see what folks that are way smarter than me about this stuff think.

I can't CAD my way out of a wet paper bag, otherwise I'd give it a shot.

elcrni, 2023-01-11:
There are plans on this but no ETA yet.

tommy536, 2023-01-22:
That would be great!

suizhenmin, 2023-01-23:
So excited to finally be able to wait

AmpereSi, 2023-02-10:
I have spent countless hours today looking for this very information, and saying I cant believe someone hasn't created a better solution for the filament sensor than leaving it all open.  I actually just hacked apart this build to see if I could figure out what to do.  I really need to get learning cad.  

Thanks for thinking about making this, this version thus far is the nicest.

Gabbajoe, 2023-02-13:
Is there away to support you to speed up the release with a filament sensor support? :D

hazza64, 2023-04-05:
I had a hack at designing this and came away a couple of points:

- There's no good way of just cutting out a clearance for the filament sensor and keeping all the mount points for things like the PCB bracket, cable chain bracket, door hinge etc, so you'd need to effectively incorporate the geometry for the filament sensor housing (the grey printed part that comes with the kit) into the read_body part
- The orientation of the print for the rear_body means massive overhangs, like 'too big to accurately print', if you just merge the filament sensor geometry with the rear_body.
- The only solution I could think up would be breaking the rear_body into two parts; one that is effectively the rear_body with a mating interface for a second part. That second part would contain the geometry for the pcb, ball bearing etc. It's all real tight in there and I gave up at this point and put it in the 'too hard' basket.

I'd be very interested in seeing what others come up with, though!

dkowis, 2023-04-12:
https://cults3d.com/en/3d-model/tool/orbiter-filament-sensor-clockwork-2-mod

Someone has created a modification for Clockwork2 to put this on there, perhaps it can provide some inspiration :D

hazza64, 2023-04-13:
Haha that's almost exactly how it would look in my head. Thanks for sourcing this! Will have to give it a go

Hellrox, 2023-05-18:
as soon there is a mod I will build my own voron and order your mod via LayerGURU, I promise :D

Vission01, 2023-06-07:
Any timeframe as to when you might have the filament sensor add on version ready?

maroussel, 2023-06-11:
I was curious looking forward to this design as well, and I understand that @elcrni might not have the time to look into it since it's almost a new design completely. I gave it a try this morning. There is a lot of parts to be modified to make it works. I was able to overcome many of the challenges and I have a clear path to success (at least for the elements I need in my setup: PCB + Chain). Didn't liked the idea to redesign the filament sensor housing like proposed earlier(too much trouble for most and look bulky). I'm trying to keep the changes as low as possible. I don't give any ETA especially that I'm still waiting for my sensor to be delivered before I can make sure it fits well.

fryc88, 2023-07-21:
Would be perfect if someone could mix up something like that into this enclosed project:
https://www.orbiterprojects.com/orbiter-filament-sensor/

ClammyBozeman, 2024-12-07:
An easier, more clean solution is to implement a precision steel ball, as shown here: https://www.printables.com/model/501237-improved-voron-stealthburner-filament-runout-senso

Essentially it used a small steel ball, similar to a ball inside a bearing, which is activates a micro/limit switch. When the filament runs out, the filament will no longer be pushing on the ball, therefore releasing pressure from and deactivating the micro-switch sending a signal to the printer that there is no filament. 

The benefits to this it is super simple to remix in a fusion 360, its seamless, small, simple, cheap, and will use as much filament as possible before stopping.