elcrni/Voron-Mods #10: Orbiter and 5015 Fan are too close

FanSails, 2023-01-09:
1、Orbiter and 5015 Fan are too close ，that Orbiter will be squeezed to Fan during installation. Whether the clearance between them can be increased to make assembly easier.
[image]

2、Latch and body interfere. When manually loading Filament, Latch cannot be released well.
[image]

3、I like your design very much, thank you for your development

elcrni, 2023-01-11:
Thanks for the feedback.

1. This is now fixed, new files will be on GH today. 0.75mm is the distance now, although it worked just fine for me with the previous version, without any gap or issues.
2. They do interfere with SB cover and that was never in question. I dont plan to change that, first as i never ever use the lever (apart from major maintenance but the the SB cover if of the body anyway) and second as i do not want to cut the SB body for that.
3. thank you!

elcrni, 2023-01-11:
Fixed

closed 2023-01-11 by elcrni

Hellrox, 2023-06-18:
@elcrni what brand do you use ? I tried a Sunon MF50152VX-1L02C-A99 but it doesn't fit well
I'll order and try a GDSTIME probably

elcrni, 2023-06-18:
> @elcrni what brand do you use ? I tried a Sunon MF50152VX-1L02C-A99 but it doesn't fit well
> I'll order and try a GDSTIME probably

Gdstime usually.

Hellrox, 2023-06-24:
@elcrni The Gdstime dosen't fit as well, I'll contact Layerguru, maybe they printed the older version.

Heerozh, 2023-12-19:
I found that the official 3D model is wrong. For example, the front face I bought is 0.26mm thicker than the 3d mockup. The screws part of the LDO motor are 0.7mm thicker. I used your latest model and still can't fit it in. 5015 still needs about 1.5mm deeper

elcrni, 2023-12-20:
> I found that the official 3D model is wrong. For example, the front face I bought is 0.26mm thicker than the 3d mockup. The screws part of the LDO motor are 0.7mm thicker. I used your latest model and still can't fit it in. 5015 still needs about 1.5mm deeper

New, redesigned front cover is now in STL folder, deeper than initial and will accommodate any part cooling fan