elcrni/Voron-Mods #13: Mounting of the hartk two piece PCB

tommy536, 2023-01-17:
Hello, 

thanks for ur great work!

How have i to mount the hartk pcb? I am using the two pieces variant of it with CW2. But i cant find a picture of the mounted pcb. Where have i to place the pcb_mount?

tommy536, 2023-01-19:
Ok, the two piece PCB is history. I will upgrade to CAN with a EBB36. Is it correct, that the Can Board is rotated -90 degrees? At the moment, this mod is the best i have printed in the meantime! The usage will be tested in 1-2 weeks after the delivery of the CAN parts.

bjensen, 2023-01-19:
I will also use ebb36 and waiting for the last bits to arrive

elcrni, 2023-01-19:
Keep in mind EBB36 is rotated 90 degrees, this means ADXL sensor may not behave as if it was placed in intended orientation. You may need to use external ADXL to do the input shaping. Perhaps there is a way to change the orientation of the sensor in the script, i have not tried.

tommy536, 2023-01-20:
The Sensor is having a fixed rotation? I thought, the sensor grabs its positions at power on as "0,0,0". Do u know where i can find a script to rotate? Or is it be possible to add a mount in the original rotation? Maybee hard bcs. of the PG7 Mount.

elcrni, 2023-01-22:
> The Sensor is having a fixed rotation? I thought, the sensor grabs its positions at power on as "0,0,0". Do u know where i can find a script to rotate? Or is it be possible to add a mount in the original rotation? Maybee hard bcs. of the PG7 Mount.

I am not an expert for ADXL sensor but it seams that it does have fixed axes/orientation:
https://www.engineersgarage.com/adxl345-accelerometer-sensor-how-to-use/

ReXT3D, 2023-01-22:
> > The Sensor is having a fixed rotation? I thought, the sensor grabs its positions at power on as "0,0,0". Do u know where i can find a script to rotate? Or is it be possible to add a mount in the original rotation? Maybee hard bcs. of the PG7 Mount.
> 
> I am not an expert for ADXL sensor but it seams that it does have fixed axes/orientation: https://www.engineersgarage.com/adxl345-accelerometer-sensor-how-to-use/
> 
> So, i would use external ADXL for IS measurements, unfortunately i was not able to orient EBB any other way that would fit

Klipper input shaper currently uses a "sum" of all axes for the calculations.  What this means is that the orientation of the ADXL chip does not affect the results.

Dmitry prepared a PR a while ago to introduce calibration of the ADXL orientation (via a transformation matrix) but Kevin has not merged it because there is currently no need for it.

In summary, it does not matter which way the ADXL chip is oriented for IS measurements...

elcrni, 2023-01-22:
> Klipper input shaper currently uses a "sum" of all axes for the calculations. What this means is that the orientation of the ADXL chip does not affect the results.
>
> Dmitry prepared a PR a while ago [...]

Well then, thats great news! thanks for the tip :-)

ReXT3D, 2023-01-22:
> Well then, thats great news! thanks for the tip :-)

You're welcome.  And here is a link for reference: https://github.com/Klipper3d/klipper/pull/4931

elcrni, 2023-01-22:
> > Well then, thats great news! thanks for the tip :-)
> 
> You're welcome. And here is a link for reference: [Klipper3d/klipper#4931](https://github.com/Klipper3d/klipper/pull/4931)

Great! Thanks!

emersonsc, 2023-02-04:
@elcrni I am interested in the mounting of the Hartke Two Piece PCB. Is it possible with this?

tommy536, 2023-02-04:
I think this isn't possible. Bcs. the front of the orbiter 2 is having more depth to the front. Bcs of this, the SB Front is also modified in depth.

hazza64, 2023-04-05:
> I think this isn't possible. Bcs. the front of the orbiter 2 is having more depth to the front. Bcs of this, the SB Front is also modified in depth.

Correct, I have the 2 part Hartk-insired  Mellow Fly CAN board (SB2040 - [Link here](https://kb-3d.com/store/controllers-displays-drivers/793-mellow-fly-sb2040-v2-can-toolhead-pcb-for-voron-stealthburner-1674910664639.html)

I found to my dismay last night that I won't be able to use the FAN/LED second board as the Orbiter V2 is extends into the previously empty space behind he blower fan and interferes not just with the second PCB, but also the blower fan with the 'stock' Sealthburner cover, which I neglected to print as I am doing the Nozzle Cam mod.

Wish I'd known this _before_ I pulled apart my printer to build the new hotend, but serves me right to RTF-Readme.md :)

TL;DR:

There's no clearance between the orbiter and the blower fan with this design (which is fine as long as you don't have a PCB there) and you need to print the StealthBurner cover included in this design for everything to fit.