elcrni/Voron-Mods #14: Please consider changing screws to M3x25

bjensen, 2023-01-19:
Is there a reason to why M3x26 is used instead of M3x25? The 25mm screws are already used in CW2 and are much easier to source than 26mm.

elcrni, 2023-01-19:
hey, that was a typo, it is 25mm. thanks

bjensen, 2023-01-19:
Have you tested the print ? I used 25mm screws and I couldn't get them to reach, thus I made this post

elcrni, 2023-01-19:
Photos on the main page sort of show that i did test the print, heatset insert goes a tiny bit deeper and 25mm screw works just fine.

bjensen, 2023-01-19:
You are right. You just have to supply enough pressure when inserting the screw.

elcrni, 2023-01-19:
> You are right. You just have to supply enough pressure when inserting the screw.
> 
> Let's close this issue and thanks for the prompt reply ✌️

;-) thanks!

closed 2023-01-20 by bjensen