elcrni/Voron-Mods #16: Orbiter: Too loose fit?

bjensen, 2023-01-21:
As you can seee in the attached video, ev

Uploading IMG_0531.mov…

en after all the screws are in, it seams that the orbiter is able to wiggle in the mount..

tommy536, 2023-01-22:
This is not a normal behavior. Mine is absolutly solid.

bjensen, 2023-01-22:
> @ github. com>

Did you reprint the newest version?

Mvh
Brian Jensen
Nordplaner
+45 60712000

tommy536, 2023-01-22:
> Did you reprint the newest version?
> 
> 

I printed the parts first time 3 days ago.

elcrni, 2023-01-22:
> As you can seee in the attached video, it seams that the orbiter is able to wiggle in the mount..
> 
>  IMG_0531.mov

I am not sure how you got the orbiter to "wiggle". I have printed each and every iteration and tested everything, solid as a rock.

bjensen, 2023-01-22:
Ah. I thought some screws

[image]

bjensen, 2023-01-22:
I had forgotten some screws [image]

elcrni, 2023-01-22:
> I had forgotten some screws [image]

That happens. Perhaps avoid submitting an "issue" before you check everything first. thanks