elcrni/Voron-Mods #17: Simple question about PTFE tube

ShryuKt, 2023-01-23:
Hi,

I can't find the answer in the repo or in an already open question,
In the CAD, the length is 37.12mm but as I'm using a rapido and I just want to be sure.

So what length are you using for the PTFe tube between the hotend and the extruder?

Thanks you

RudyAlbertoAcostaMarmalade, 2023-03-22:
Interesting, For my Dragon HF Hotend 35mm seems to be spot on.

RudyAlbertoAcostaMarmalade, 2023-03-22:
> in the CAD the tube is too short. he doesn't go into the orbiter there.

@tome9111991 - You made me question my install  & you are indeed correct  - 41mm is right for a Dragon Hotend.  

My alignment between the Orbiter and the printed parts was slightly out, I believe due to the heated thread inserts, so I reassembled the extruder housing with PTFE tube in place first and all is well!  Thank you.

tome9111991, 2023-03-22:
yes 41 is correct.
have same issue with alignment.
with this "long" ptfe tube the filament flow is very nice.

ShryuKt, 2023-03-26:
Thank you for your answers 40,1 it is ^_^

closed 2023-06-11 by elcrni