elcrni/Voron-Mods #18: Ebb36 Adapter mit working with tap 5v pcb

Malygros, 2023-01-30:
[image]

Hey,
I got issues when installing the new ebb36 with tap pcb.

there is no manual for mounting but I think this must be the way.

when I install the ebb36 adapter it collide with tap pcb.

tommy536, 2023-01-30:
This is the wrong side. The Umbi holder is mounted on the right when u look from behind. [image]

Malygros, 2023-01-30:
> This is the wrong side. The Umbi holder is mounted on the right when u look from behind. [image]

Arg, That came never to my mind. Ty for the Info. Maybe the Readme should be updated :D

tommy536, 2023-01-30:
I think it is to use so 😀 Bcs. umbi is the counterpart of the Chains. The lower position is also good. But, in the meantime i am using a another solution with the standard EBB Mounts for a Orbiter. At the Moment i cannot say if the hight is to much to the top panel.

Malygros, 2023-01-31:
It works fine. Just installed everything. Will there be a Cover for the cables from EBB36 soon? PCB is not working.

elcrni, 2023-06-11:
> It works fine. Just installed everything. Will there be a Cover for the cables from EBB36 soon? PCB is not working.

Correct cover for the EBB36 has been added to STL folder.

closed 2023-06-11 by elcrni