elcrni/Voron-Mods #19: Rapido UHF

Flapp66, 2023-02-01:
Hi

Is it possible that you could modify this to fit the Rapido UHF?

bjensen, 2023-02-01:
Hi @Flapp66, 

I also have an UHF too , so I opened up https://github.com/elcrni/Voron-Mods/blob/main/Orbiter_2.0_SB_CW2_Enclosed/STL/SB_main_Body.stl and https://github.com/bythorsthunder/VoronUsers/blob/1917f585ad662b4bfbe48eae9bf47e91c1254f9c/printer_mods/bythorsthunder/Stealthburner_Rapido_Uhf/STL_FILES/%5Ba%5D_Stealth_Rapido_UHF_Front.stl in Fusion360 and even though Im a newbie in it, it seams that the only difference was the extended bottom, so Im using this mod with the uhf front.

@elcrni please correct me if Im wrong.

elcrni, 2023-02-01:
hey, no, sb body was changed to have the part cooling fan deeper/more to the front. nothin else changed

bjensen, 2023-03-28:
@Flapp66 do you still need it modified? I have the STL for it

elcrni, 2023-06-11:
Rapido UHF support has been added! Thanks

closed 2023-06-11 by elcrni