elcrni/Voron-Mods #21: Can't mount EBB36 and Cover Latch

erlis, 2023-03-05:
I saw two version of the cover and I've tried both, but I can't use it with the EBB, check picture below

[image]

I can't find the cover with that on the files provided, I exported the cover from the CAD and I'm printing that to test one more time. I think I'm missing something because a lot of people are using the EBB board

erlis, 2023-03-06:
UPDATE: After printing the file I exported from the CAD, the same one I'm pointing here, that cover works perfect with the EBB, I think it needs to be added on the STLs as COVER-FOR-EBB

elcrni, 2023-05-05:
EBB cover has been added as the STL to the main repository. many thanks

closed 2023-05-05 by elcrni