elcrni/Voron-Mods #29: Load filament

yardi83, 2023-12-31:
Hi!!

Thanks for your design. Great work so far!
Installed everything yesterday and maybe it`s just me beeing dumb but what is the procedure for loading filament into the orbiter?

Remove PTFE Tube, open PCB Cover, untighten the tensioner screw on orbiter und then?
I used the orbiter 2 on other machines and there i had to pull the lever back to get filament in and that isn`t really working for me with this mod here. Everything is so tight with the stealburner that i can`t move the lever. Stealthburner is blocking the lever.

Or should untighten the tensioner screw and pushing the filament in be working? Because for me it isn`t :(

elcrni, 2023-12-31:
hey there, thanks for the comment.

the easiest way to load/unload filament by far is the load/unload macro. with the hotend preheated you would just hit UNLOAD command and the filament would unload/come out... for loading just feed the filament untill it touches gears and hit LOAD command, the filament will load and purge previous...
there are different version of load/unload macros available depending on your firmware but they should all do the same thing.

yardi83, 2023-12-31:
Thanks for your reply

Seems like i am more of a "doing IT manually" Type of guy :)

Seems workable. Setting up a Load Macro in Klipper shouldn't be a big Problem.

I used one but with a bowden Setup. With DD i usually do it by hand

closed 2023-12-31 by yardi83