elcrni/Voron-Mods #30: Doesn't work LDO Orbiter 2

d4mer, 2024-04-03:
I can confirm that an LDO Orbiter 2 and GDStime fan doesn't fit with with this toolhead. I printed the deep version and it still doesn't work
[image]
[image]
[image]
[image]

closed 2024-04-03 by d4mer

ssbb, 2024-09-27:
@d4mer Just wondering what changed for you? I think it's not deep enough in my case too (gdstime fan as well).

I been using the normal version for some time and recently printed the deep one. Things are improved but not that much. When I put the screws in - orbiter just pushes the fan and it's misaligned.

ssbb, 2024-09-28:
Answering my own question - it seems like I didn't pushed the fan till the end. I had to sand it down quite a lot so it fits on the lower cutout.